librelane/librelane #694: KLayout Streamout options

mole99, 2025-06-13:
KLayout offers several options for resolving cell conflicts.

I made use of this feature in [Greyhound](https://github.com/mole99/greyhound-ihp/tree/main) to reduce the GDS file size drastically:

```
# Save on file space by sharing same cells
PRIMARY_GDSII_STREAMOUT_TOOL: klayout
KLAYOUT_CONFLICT_RESOLUTION: SkipNewCell
```

I intend to upstream this to LibreLane soon.

smunaut, 2025-07-07:
Looking fwd to that since this is one of the patch I need to carry in a fork to set `RenameCell` 😅

mole99, 2025-07-07:
You can find the patch in my OpenLane 2 `greyhound` branch: https://github.com/mole99/openlane2/commit/c699ddbb4a36289ee9ed96a499a5db2341d20402
Will make sure to merge this into `librelane/ihp` soon :)

@donn Is `KLAYOUT_CONFLICT_RESOLUTION` ok as a variable name?

smunaut, 2025-07-14:
That's why RenameCell is the default.

But in @mole99 case, when you assemble in a top level a lot of macro (hundreds) build out of standard cells, if you use Rename, then each macro ends up with its own private copy of the standard cells and that yields to much larger files. But because he knows how those macro are built and they are guaranteeds to use the same source of the standard cells, using "Skip" is a valid and safe option

stefanottili, 2025-07-14:
If you use strm2gds, you could just load the macro def and the stdcell lef/gds.
This way the def components will be unresolved references, resolved by the stdcell gds.

mole99, 2025-07-15:
Hi Stefan, strm2gds could be an alternative streamout step in LibreLane. Let us know if you're up for the challenge.