librelane/librelane #695: Don't touch directive violated

mole99, 2025-06-13:
LibreLane allows to mark nets as "don't touch" using `RSZ_DONT_TOUCH_RX`.

Unfortunately, the time-driven placement does not respect this.
Therefore, I had to disable time-driven placement in [Greyhound](https://github.com/mole99/greyhound-ihp/tree/main):

```
# Time driven placement ignores don't touch directives
# Therefore only enable routability driven
PL_TIME_DRIVEN: False
PL_ROUTABILITY_DRIVEN: True
```

donn, 2025-06-13:
Related: https://github.com/The-OpenROAD-Project/OpenROAD/issues/6643

I think a quick workaround is to set `PL_KEEP_RESIZE_BELOW_OVERFLOW` to `0`.

An actual solution would be to add invocations of `set_dont_touch_objects` and `unset_dont_touch_objects` to `gpl.tcl` similar to the resizer scripts. Would you mind testing that?

donn, 2025-06-15:
So I slept on it and:

The reason we unset it is that this variable is for the resizer step specifically. Because setting dont touch during placement will mean that it won't place the cells in question- it's not just about resizing!

I think a good solution would be to set `PL_KEEP_RESIZE_BELOW_OVERFLOW` to 0 by default. Sure, the metrics are going to be a bit worse. But it's less surprising behavior for the end-users and they can still opt-in to the behavior if they know the risks.