librelane/librelane #730: build without NIX

jalcim, 2025-07-25:
It is possible to build without this sh*t called nix ?
It is possible to use pip instead ?

mole99, 2025-07-26:
As stated in the documentation you will need to provide the tools yourself.

Also note:

> However, as the versions will likely not match those packaged with LibreLane, some incompatibilities may arise, and we will not be able to support them.

We chose Nix because it allows us to easily bundle LibreLane with all the tools whilst managing dependencies. If you don't want to use Nix, than that's your choice. However, don't call Nix shit just because it doesn't work for you.

The other alternative is to use the Docker container.

donn, 2025-07-26:
As the documentation outlines, you are free to either use Docker or build the accompanying tools yourself at your own risk.

Also, kindly refer to the [FOSSi Foundation Code of Conduct](https://fossi-foundation.org/code-of-conduct) for how you choose to engage with maintainers.

closed 2025-07-26 by donn