librelane/librelane #774: Instructions for installing nix+librelane as "user install" without root?

mithro, 2025-09-20:
Is there a way to install librelane using nix without needed root?

mithro, 2025-09-20:
The instructions @ https://nixos.wiki/wiki/Nix_Installation_Guide#Installing_without_root_permissions seem to cover how to use nix without root.

mithro, 2025-09-20:
It seems the instructions for librelane @ https://librelane.readthedocs.io/en/latest/getting_started/common/nix_installation/installation_linux.html#installing-nix are using ["Determinate Nix"](https://docs.determinate.systems/determinate-nix) rather than stock NixOS?

mole99, 2025-09-22:
We use the determinate Nix installer to install upstream Nix, as this has worked best.
However, this may change, see here: https://github.com/librelane/librelane/pull/770